kmicki/SteamDeckGyroDSU #1: Doesn't work in botw

BUBBL33, 2022-05-06:
I downloaded it, done all the steps, it is selected in cemu but when I play botw it doesn't work.

kmicki, 2022-05-06:
Cemu running on Deck or separate PC?

tom300z, 2022-05-06:
@kmicki 
I think the problem occurs when cemu is closed and the cemuhook server stops.

adding a 1s sleep after the join call fixed it for me, though this method is probably not ideal (i haven't really programmed c++ so far):
https://github.com/kmicki/SteamDeckGyroDSU/blob/74dbd094baa283edff2c6f55a78bb6047f2fe1d6/src/cemuhook/cemuhookserver.cpp line 204-212:
```
                    if(sendThread.get() != nullptr)
                    {
                        std::cout << "Cemuhook Server: No packet from client for some time. Stop sending data." << std::endl;
                        stopSending = true;
                        sendThread.get()->join();
                        std::this_thread::sleep_for(std::chrono::seconds(1)); // <----------------------------
                        sendThread.reset();
                    }
                    sendTimeout = 0;
```

tom300z, 2022-05-06:
If CEMU is already open while the service is started the server will never work. 
Completely restarting the deck fixed it for me

kmicki, 2022-05-06:
So far I've found that sometimes HidDevReader::readTask will start but it will get stuck and no frames will be read. There was a mechanism to detect this situation and nuke the entire process, but it was bugged. I fixed it in the [release v1.6](https://github.com/kmicki/SteamDeckGyroDSU/releases/tag/v1.6). The service is configured so that the server will be restarted when that happens. After a short while it should work again.

I will try to debug and find an actual cause.

kmicki, 2022-05-06:
> If CEMU is already open while the service is started the server will never work. Completely restarting the deck fixed it for me

That is a strange behavior. Do you mean that when it happens once then restart is required? I mean, you start Cemu, start the service, it doesn't work and then when you close Cemu, restart the service and open Cemu again it will not work?

Cemuhook should poll a server for controller data periodically. It should work in those situations. I haven't encountered the problem with Cemu on a separate PC. When I have Cemu on my Deck finally, I will test that.

tom300z, 2022-05-07:
> do you mean that when it happens once then restart is required? I mean, you start Cemu, start the service, it doesn't work and then when you close Cemu, restart the service and open Cemu again it will not work?

Restarting the service (while CEMU is closed) also works, a reboot is just an easier way to do that for peaple who don't know systemd and stuff.

I also observed that the scan period never gets updated after the "Scan period initial :" log message when the service is started while cemu is running. This also happens when you restart Cemu without restarting the "tainted" service. 

When the service is started correctly (before cemu) i get about 10 "Changed scan period to: xxxx" messages everytime i start cemu

kmicki, 2022-05-07:
Ok, thank you for information. I just confirmed again that it doesn't happen with Cemu on another PC. I can restart the service while Cemu is running, stop it and start after a long while and Cemu picks it up.

The situation you're describing happens sporadically and after service restart (automatic now after a few seconds) it works again.

I will install Cemu on Deck and it might help with finding the root cause of the task freezing.

BUBBL33, 2022-05-07:
I have installed your software and cemu on the deck itself. In desktop mode I can go to cemu and do step 4 (gamepad motion source-dsu1-by slot) and it will select it. Then once I go into gaming mode and start botw the motion controls don't work.

also, I'm very new to linux so if there are any logs to dump or files to check or whatever please let me know.

kmicki, 2022-05-07:
@BUBBL33 : please execute this command in Konsole:

    systemctl --user status sdgyrodsu.service

And copy the output here.

In Cemu DSU1 option would have zeroed MAC in parenthesis like: _DSU1 (00:00:00:00:00:00)_ if it detects the server. Does it?
Also, did you put cemuhook.dll into Cemu folder? Was cemuhook.ini generated? If so, try inserting:

    server IP=127.0.0.1

under `[Input]`.

Just to be sure: how in game do you check motion controls. If you do that by aiming with a bow: do you have motion aiming enabled in the settings inside the game?

BUBBL33, 2022-05-07:
Here is the Konsole,

(127)(deck@steamdeck ~)$ systemctl --user status sdgyrodsu.service
● sdgyrodsu.service - Steam Deck Gyro DSU Server
     Loaded: loaded (/home/deck/.config/systemd/user/sdgyrodsu.service; enabled; vendor preset: enable>
     Active: active (running) since Fri 2022-05-06 22:12:35 CEST; 11h ago
   Main PID: 1073 (sdgyrodsu)
      Tasks: 2 (limit: 17718)
     Memory: 1.2M
        CPU: 2.191s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/sdgyrodsu.service
             └─1073 /home/deck/sdgyrodsu/sdgyrodsu

May 06 22:14:35 steamdeck sdgyrodsu[1073]: Changed scan period to : 3926 us
May 06 22:14:35 steamdeck sdgyrodsu[1073]: Changed scan period to : 3927 us
May 06 22:14:35 steamdeck sdgyrodsu[1073]: Changed scan period to : 3928 us
May 06 22:14:35 steamdeck sdgyrodsu[1073]: Changed scan period to : 3929 us
May 06 22:14:35 steamdeck sdgyrodsu[1073]: Changed scan period to : 3930 us
May 06 22:15:00 steamdeck sdgyrodsu[1073]: Cemuhook Server: No packet from client for some time. Stop >
May 06 22:15:00 steamdeck sdgyrodsu[1073]: Cemuhook Server: Initiating frame grab stop.
May 06 22:15:00 steamdeck sdgyrodsu[1073]: HidDevReader: Attempting to stop frame grab.
May 06 22:15:01 steamdeck sdgyrodsu[1073]: HidDevReader: Stopped frame grab.
May 06 22:15:01 steamdeck sdgyrodsu[1073]: Cemuhook Server: Stop sending controller data.
lines 1-20/20 (END)

In cemu it says DSU1 USB DS4 00:00:00:00:00:00

the cemuhook dll and ini are inside the cemu folder. I added the server ip to cemuhook.ini

also installed the newest release

when start botw in desktop mode or gaming mode and go to the settings I cannot select motion controls.

kmicki, 2022-05-07:
@BUBBL33 :

> when start botw in desktop mode or gaming mode and go to the settings I cannot select motion controls.

Oh then it's clear: in Cemu in input settings you probably have WiiU Pro Controller selected. Change it to WiiU Gamepad. Pro Controller doesn't have motion built in so game just ignores it.

BUBBL33, 2022-05-07:
Well that was part of it at least. I can select motion control now, but when I grab a bow and try to use motion it still doesn't work.

kmicki, 2022-05-07:
Okay, please do the Konsole thing when Cemu is running.

BUBBL33, 2022-05-07:
sdgyrodsu.service - Steam Deck Gyro DSU Server
     Loaded: loaded (/home/deck/.config/systemd/user/sdgyrodsu.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-05-07 10:44:11 CEST; 17min ago
   Main PID: 1069 (sdgyrodsu)
      Tasks: 6 (limit: 17718)
     Memory: 1.7M
        CPU: 1min 47.552s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/sdgyrodsu.service
             └─1069 /home/deck/sdgyrodsu/sdgyrodsu

May 07 10:44:11 steamdeck sdgyrodsu[1069]: Cemuhook Server: Initializing.
May 07 10:44:11 steamdeck sdgyrodsu[1069]: Cemuhook Server: Initialized.
May 07 10:44:11 steamdeck sdgyrodsu[1069]: Cemuhook Server: Start listening for client.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: Cemuhook Server: Client subscribed to data events.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: Cemuhook Server: Initiaiting frame grab start.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: HidDevReader: Attempting to start frame grab.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: HidDevReader: Started frame grab.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: Cemuhook Server: Start sending controller data.
May 07 10:45:37 steamdeck sdgyrodsu[1069]: Scan period initial : 3900 us
May 07 10:49:36 steamdeck sdgyrodsu[1069]: Cemuhook Server: No packet from client for some time. Stop sending data.
(deck@steamdeck ~)$

kmicki, 2022-05-07:
Okay, it might be the problem that @tom300z and for some reason abort is not working.

Stop Cemu, then execute:

    systemctl --user restart sdgyrodsu.service

Then wait a few seconds and start Cemu again.

BUBBL33, 2022-05-07:
Yes this works! It's pretty good. Can I change the gyro sens and stuff in steam or cemu settings?

kmicki, 2022-05-07:
No, unfortunately not unless the game provides the setting.

kmicki, 2022-05-07:
@tom300z : I found the reason for the freeze. It happens at random just after the client connects to the server (maybe more often when client subscribes very quickly after service startup and form observations definitely more often when client is on the Deck and not on the separate PC in the network). 

[Release v1.8](https://github.com/kmicki/SteamDeckGyroDSU/releases/tag/v1.8) fixes the problem.

I'm closing the issue.

closed 2022-05-07 by kmicki