kmicki/SteamDeckGyroDSU #2: Add Zip Update Script in the Zip File

IBNobody, 2022-05-07:
I wrote an update script for you to include in the zip file. Note that it just updates the bin and service, not anything else.

```
@echo off
echo "Grabbing latest .zip file..."
curl -L -O https://github.com/kmicki/SteamDeckGyroDSU/releases/latest/download/SteamDeckGyroDSUSetup.zip
unzip -o SteamDeckGyroDSUSetup.zip -d /home/deck
echo "Stopping service..."
systemctl --user stop sdgyrodsu.service
systemctl --user disable sdgyrodsu.service
echo "Copying files..."
rm $HOME/sdgyrodsu/sdgyrodsu
cp sdgyrodsu $HOME/sdgyrodsu/
chmod +x $HOME/sdgyrodsu/sdgyrodsu
echo "Installing service..."
rm $HOME/.config/systemd/user/sdgyrodsu.service
cp sdgyrodsu.service $HOME/.config/systemd/user/
systemctl --user enable --now sdgyrodsu.service
echo "Installation done."
systemctl --user status sdgyrodsu.service
```

kmicki, 2022-05-07:
Would you like to issue pull request to be listed as contributor to this change? The files in /pkg/ folder in the repository are going into zip packages together with a built binary.

IBNobody, 2022-05-07:
Yeah, I just created a fork and PR for it. I didn't want to do a PR initially because I wasn't sure how you were building the .zip. Your message clarified it. Thanks! Here's the PR: #3