kmicki/SteamDeckGyroDSU #9: Yuzu compatibility

baggyg, 2022-05-10:
All working in Cemu -> excellent. Thank you so much! 

The server itself seems to be detected in Yuzu 127.0.0.1 Port 26760. The test shows data is being received. However this never makes it through to the game it seems. Are these the correct settings?

closed 2022-05-10 by baggyg

KrisCris, 2022-09-03:
So how exactly did you solve the issue? I installed everything via emudeck and ran into the same problem, even folks in emudeck discord have no idea.

baggyg, 2022-09-07:
Which problem? 
The related issue is that by default on steam deck its hard to see the motion button. 
I connected to a monitor to see it:
[image]

https://github.com/yuzu-emu/yuzu/issues/8206