kmicki/SteamDeckGyroDSU #10: Installation Failed per ssh

nille02, 2022-05-12:
I did try to install it over ssh and i git a error for it.

```
(1)(deck@steamdeck SteamDeckGyroDSUSetup)$ ./install.sh
Checking 'usbaccess' user group...
'usbaccess' group exits.
Checking if current user is in an 'usbaccess' group...
User is in an 'usbaccess' group.
Checking if there are permissions for 'usbaccess' group in place...
Permissions are present.
Stopping the service if it's running...
Copying binary...
Binary copied.
Installing service...
Failed to connect to bus: No medium found
Failed enabling the service.
```

after that i did try it direct on the deck in a terminal an that works.

kmicki, 2022-05-12:
There are a lot of missing files in packages installed on deck's system (removed by Valve probably for some reason). Maybe it is the case for `systemd` package also.

closed 2022-05-12 by kmicki

nille02, 2022-05-12:
Nothing changed. Steam OS is on the beta. But its not important, because installing it directly from the desktop works. after restarts the service is still started and works well.

nille02, 2022-05-12:
I guess i found the issue, my sshd did not use PAM, so systemd did not set the required variables.