kmicki/SteamDeckGyroDSU #15: SDGyroDSU Stopped working

bpiccolo88, 2022-05-27:
I've tried uninstalling and restarting the steam deck but it doesn't work in game. I previously had it working but reinstalled cemu and it stopped working ever since. Gyro is working fine otherwise. Attached is the console output when cemu is running.
[sdgyrodsu.log](https://github.com/kmicki/SteamDeckGyroDSU/files/8788575/sdgyrodsu.log)

kmicki, 2022-05-27:
From this short log it looks like there are multiple clients trying to read data. Every time new client tries to read data, the current client is dropped. Please generate longer logfile with command:

    journalctl --user -u sdgyrodsu > sdgyrodsu.log

and attach it here or check if there are multiple messages:

    Server: Request to subscribe from new client. Dropping current client.

Which version of Cemu do you use? Maybe there are multiple instances running fir some reason?

There is handling multiple clients in plans (#13).

bpiccolo88, 2022-05-27:
@kmicki , attached is the log running the journalctl command. I do see  `Server: Request to subscribe from new client. Dropping current client.` quite a bit. I am using cemu 1.26.2f. I am only running Cemu once, not sure what else would be trying to use the dsu service
[sdgyrodsu (2).log](https://github.com/kmicki/SteamDeckGyroDSU/files/8788905/sdgyrodsu.2.log)
.

kmicki, 2022-05-27:
It looks like Cemu tries to connect with 2 clients. My guess is you have motion input enabled via cemuhook (Options->Gamepad Motion Source->DSU1->By Slot) and in native Cemu support (Options->Input settings). In Input Settings, try selecting DSU Controller and removing it using - (minus) button.

bpiccolo88, 2022-05-27:
@kmicki - You were right. I use Emudeck for managing emulators on the deck and it added a DSU controller. I removed it, restarted the deck, and now motion controls are working again. Thanks!!

closed 2022-05-28 by kmicki