kmicki/SteamDeckGyroDSU #16: Reading from hiddev file failed.

humanoidcreature, 2022-05-29:
Hi kmicki,

I'm trying to make the gyro work on steam deck, but I'm facing some issues.
Latest version of the repo is installed, the correct Wii U controller is selected in Cemu, I removed the DSU controller input what Emudeck installed. I think I checked everything what I can, but still no gyro in botw.
Here's the status while cemu is running:
```
● sdgyrodsu.service - Steam Deck Gyro DSU Server
     Loaded: loaded (/home/deck/.config/systemd/user/sdgyrodsu.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-05-29 20:04:07 CEST; 3s ago
   Main PID: 17089 (sdgyrodsu)
      Tasks: 6 (limit: 17764)
     Memory: 708.0K
        CPU: 329ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/sdgyrodsu.service
             └─17089 /home/deck/sdgyrodsu/sdgyrodsu

May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reading from hiddev file started in the middle of t>
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reopening hiddev file...
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reading from hiddev file started in the middle of t>
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reopening hiddev file...
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reading from hiddev file started in the middle of t>
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reopening hiddev file...
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reading from hiddev file started in the middle of t>
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reopening hiddev file...
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reading from hiddev file started in the middle of t>
May 29 20:04:10 steamdeck sdgyrodsu[17089]: HidDevReader: Reopening hiddev file...
```

And here's the log:
[sdgyrodsu.log](https://github.com/kmicki/SteamDeckGyroDSU/files/8793973/sdgyrodsu.log)

humanoidcreature, 2022-05-30:
[hiddevlog.zip](https://github.com/kmicki/SteamDeckGyroDSU/files/8797757/hiddevlog.zip)

I couldn't upload .hex file here directly, so I put it into a zip file.

kmicki, 2022-05-30:
For some reason your hiddev file behaves differently than mine. I hoped such things will be consistent between one Deck to the other.

I would like to prepare a test version with a fix for you to test - are you okay with it?

humanoidcreature, 2022-05-30:
Yes, sure. I'm up to it.

kmicki, 2022-05-30:
[SteamDeckGyroDSUSetup.zip](https://github.com/kmicki/SteamDeckGyroDSU/files/8798561/SteamDeckGyroDSUSetup.zip)

Above is the link to the test binary package. Enter following commands to install it on Deck:

    cd
    wget https://github.com/kmicki/SteamDeckGyroDSU/files/8798561/SteamDeckGyroDSUSetup.zip
    unzip -o SteamDeckGyroDSUSetup.zip
    cd SteamDeckGyroDSUSetup
    ./install.sh

Please check if that works for you.

humanoidcreature, 2022-05-30:
Wow it's working, really amazing. You handled it really fast! :)

By reading the status a few warnings still present, but I **didn't notice** any hiccups with the bow:
```
● sdgyrodsu.service - Steam Deck Gyro DSU Server
     Loaded: loaded (/home/deck/.config/systemd/user/sdgyrodsu.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-05-30 17:21:00 CEST; 2min 57s ago
   Main PID: 36363 (sdgyrodsu)
      Tasks: 6 (limit: 17764)
     Memory: 584.0K
        CPU: 9.695s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/sdgyrodsu.service
             └─36363 /home/deck/sdgyrodsu/sdgyrodsu

May 30 17:23:25 steamdeck sdgyrodsu[36363]: HidDevReader: Reading from hiddev file failed.
May 30 17:23:25 steamdeck sdgyrodsu[36363]: HidDevReader: Reopening hiddev file...
May 30 17:23:25 steamdeck sdgyrodsu[36363]: HidDevReader: Missed 1 frames.
May 30 17:23:25 steamdeck sdgyrodsu[36363]: HidDevReader: Replicating missing frames.
May 30 17:23:39 steamdeck sdgyrodsu[36363]: HidDevReader: Missed 32 frames.
May 30 17:23:39 steamdeck sdgyrodsu[36363]: HidDevReader: Changed scan period to : 3960 us
May 30 17:23:39 steamdeck sdgyrodsu[36363]: HidDevReader: Replicating missing frames.
May 30 17:23:53 steamdeck sdgyrodsu[36363]: HidDevReader: Missed 32 frames.
May 30 17:23:53 steamdeck sdgyrodsu[36363]: HidDevReader: Changed scan period to : 3959 us
May 30 17:23:53 steamdeck sdgyrodsu[36363]: HidDevReader: Replicating missing frames.
```
Here's the log:
[sdgyrodsu.log](https://github.com/kmicki/SteamDeckGyroDSU/files/8799477/sdgyrodsu.log)

kmicki, 2022-05-30:
Great, thank you for testing. I see in the logs that every now and then there is a problem with reading the hiddev file for a few seconds, but missing frame count is not high, so they get replicated and you do not notice hiccups.
Other warnings are a normal thing in the current version.

I still don't understand why the data looks different on your machine. Do you have Steam started or not during testing?

humanoidcreature, 2022-05-30:
Steam is running. I need it to be able to run Cemu through Proton.
What's the difference exactly?
Can it be related to character coding? (I'd be surprised, but who knows...) But anyway, only the Steam virtual keyboard is hungarian, everything else is english.

Crono141, 2022-05-30:
not to hijack this issue, but how did you go about removing emudeck version of this? I'm getting some weird behavior as well and I'm trying to troubleshoot and try this test version. Thanks.

kmicki, 2022-05-30:
@Crono141 when you run the commands in the other comment, the test version will replace the one installed by emudeck. See also #15 - another issue that happens when using Cemu from Emudeck.

Crono141, 2022-05-30:
thanks. This fixed me.

Oktavian76, 2022-06-01:
I just tried what you suggested. First I switched back to the official version oft DSU. Then started PadTest on another PC and connected it to the Deck. This works fine, I can see the movement. 

Then I started Cemu again. GamePadMotionSource is set to DSU1 and "by slot". In Input Settings the controller is "Controller 1 [XInput]", under Settings no "Use motion", it is grayed out. No other controller selectable. 

The Cemu I use is not from Emudeck but directly downloaded from their page and manually added to Steam.

[dsu.log](https://github.com/kmicki/SteamDeckGyroDSU/files/8812528/dsu.log)

Oktavian76, 2022-06-01:
Just doublechecked, "Emulated Controller" is set to "Wii U GamePad"

Oktavian76, 2022-06-01:
Yes, sometimes it is so easy. Before installing your software I use gyro from the SteamDeck to help aiming, so aiming with gyro in BOTW was disabled. Now after enableing it also in the game it works fine. Thank you for your help.

Berinkton, 2022-06-06:
@kmicki Thx its fixed 😅