kmicki/SteamDeckGyroDSU #24: Steam Deck GyroDSU horizontal axis limited input in Cemu

CBrucePerkins, 2022-09-11:
Hey I installed DSU as described in the documentation and have Cemu hook running. Native gyro controls seem to be working in BoTW for the Y-axis without issue. The problem is the X-axis depending on how I hold the deck. If I hold it upright with the screen facing me, gyro controls don’t really register when I rotate from side to side. If I lay the deck flat and repeat the movements it works as it should. 

If I enable gyro on right pad touch, the above experiments work under all conditions, so I’m inclined to think there is some input that isn’t being registered by DSU?

I’m running Cemu from EmuDeck basic install.

kmicki, 2022-09-11:
Hi, gyro on right pad touch is emulating right stick movement. It's completely different input passed to Cemu.

The issue is caused by either BOTW game specifically or how cemuhook/Cemu calculates quaternion. 

If you want it to work properly, keep your Deck lying horizontally on the flat surface when you start a game. When the game is loaded you can pick up the Deck and gyro should be correct.

CBrucePerkins, 2022-09-11:
> Hi, gyro on right pad touch is emulating right stick movement. It's completely different input passed to Cemu.
> 
> The issue is caused by either BOTW game specifically or how cemuhook/Cemu calculates quaternion.
> 
> If you want it to work properly, keep your Deck lying horizontally on a flat surface when you start Cemu and the game. When the game is loaded you can pick up the Deck and gyro should be correct.

Thanks for the reply. I wasn’t talking about the right touch pad movement input, but talking about the Steam interface gyro control options. So I’ve enabled gyro using DSU, and I’ve also enabled gyro through steam by pressing the Steam button, enabling gyro by touching the right touch pad. The two weren’t behaving the same in terms of what input was getting picked up. 

I tried starting the game with the Deck flat, and that solved the issue. 

Is it necessary to start the game with the deck lying flat every time to calibrate? Like I said, that’s not necessary while using Steams gyro controls, which lead me to believe that there was some missing input.

kmicki, 2022-09-11:
> I wasn’t talking about the right touch pad movement input, but talking about the Steam interface gyro control options.

I know what you mean. But that's how it works, Steam can only generate XInput signals or mouse/keyboard. There's no way to pass native gyro signal to Cemu with those. Look into controller layout and you'll see that your gyro is set to 'as joystick' or 'as mouse' . That means that when you move your Deck, Steam stimulates movement of right joystick or mouse.

Go to a shrine in BOTW with a gyro puzzle (like the one where you have to balance a ball on a board) and you'll see that the gyro control from Steam will not work at all for that puzzle.

SteamDeckGyroDSU+cemuhook is used to relay sensor readings to the game. It cannot control what the game does with it. And, for some reason, BOTW behaves like that.

CBrucePerkins, 2022-09-11:
Ah I see. In that case things are working properly and I’ll close the issue. I hadn’t read through everything, but is this mentioned in the documentation (that Cemu should be started with the Deck laying flat)?

closed 2022-09-11 by CBrucePerkins