kmicki/SteamDeckGyroDSU #25: FeatureRequest: Motion support for multiple devices

KS-HTK, 2022-09-16:
I am trying to use the Decks gyro and the gyro of a connected Steam Controller.
I would love to see support for ble connected controllers.
There currently does not seem to be a way to use the gyro without mapping it to a Joystick for the Steam Controller.

cydenyliu, 2022-09-19:
Second this. After the installation of this DSU, which works perfectly, the old steam-gyro-for-cemuhook gives error all the time.

kmicki, 2022-09-19:
This tool is specifically for Steam Deck's gyro. Handling Steam Controller's gyro is out of scope for this project.

steam-gyro-for-cemuhook shows error probably because it uses the same UDP port. Looking at it's github page it should be configurable there, so setting a separate port should work. I will check it as soon as I am able to.

KS-HTK, 2022-09-19:
I know of, and had tested, steam-gyro-for-cemuhook, but the issue with that is it seems to only work when using cemu to enable the controller. I could not get it to recognize the controller for use with other emulators.

And yes it required configuring a different port. But that is not why I suggested supporting it here.

The steam-gyro-for-cemuhook repository is archived, so it will not recive any updates going forward. And eventhough it has some forks, most of them are far behind. That is why I opened this request.

I do not know alot about the cemuhook or how it works, but it seems that the one there is specifically for cemu, this project can be used in multiple emulators. I am using this for yuzu for example. And while yuzu can find the server of steam-gyro-for-cemuhook under the new port, it is not reciving any motion data. The interface of that tool is also not showing any input from the controller. So I am guessing it is missing a link which cemu provides.

If you do not want to support other divices I can understand that, in that case you may want to convert this to a discussion to maybe figure out how to get it working and or or close this as not planed.

kmicki, 2022-09-19:
Hm... Maybe steam-gyro-for-cemuhook is missing the handling of the "request version" telegram. It is the only thing that cemuhook doesn't use compared to e.g. built-in DSU handling in Cemu.

If I'm going to provide Steam Controller support, it will probably not be within this project because of lack of time. The easiest way will be to fork this repo and modify code to handle Steam Controller instead of Steam Deck. 

But, it will not be possible to use them both at the same time. It will be possible to make simple script to switch between them though. I am not completely sure if that's something that would satisfy the need for Steam Controller support. Would it?

KS-HTK, 2022-09-21:
I am guessing that it is a little more, as the Steam Controller is a Bluetooth Low Energy device and probably is not constantly sending gyro data if not configured to do so. But this is just me guessing.

Having 2 services is not a problem, I figured that would be required if using both of them anyway.

Not being able to use them at the same time is not very helpful, as that is what I am trying to achieve. Maybe it could be possible to change the port on one of them. Through environment variable for example?

Anyway having it working at all would be a great step in the right direction, figuring out how to run it together can follow later.

I will close this here as you stated it is not going to happen in this repo.
Thank You.

closed 2022-09-21 by KS-HTK

kmicki, 2022-09-21:
Making port configurable via config file is a planned feature.

Does yuzu allow to choose a port? And to use 2 sources at the same time? In Cemu you can only use one, so running them at different ports would not help anyway.

KS-HTK, 2022-09-21:
I yuzu I can add as many motion sources as I wish. But since only one is currently delivering motion data I cannot say if it works using multiple simultaneously, but since the Switch Joycon settings have to motion sources themselves I guess it will.