kmicki/SteamDeckGyroDSU #26: Gather data only when connected to save CPU.

subworx, 2022-10-04:
Currently SDGDSU runs permanently in the background, causing processor load even if I don't run any emulators connecting to it.
I resorted to manually running scripts that stop/start the daemon.

Would it instead be possible to have it "sleep" until some client connects to the daemon, and only then start collecting/processing data?
I'm sure this would have quite some impact on the battery life.

Thank you :)

kmicki, 2022-10-04:
This functionality is in SteamDeckGyroDSU from the start.

`sdgyrodsu` stops sending data to client after a few seconds (around 6s) after last request for data. It has to be done that way because DSU interface is over UDP - there is no other indication of client disconnecting.

If there are no clients connected, grabbing and processing stops.

When grabbing and processing is stopped, the only thing `sdgyrodsu` does is wait for a UDP packet.

You can see it in Process  Monitor - when emulator is connected `sdgyrodsu` process consumes around 1.1% of CPU time. A few seconds after emulator is closed, `sdgyrodsu` usage drops to none. I have just checked it on my Deck with the most recent version of `sdgyrodsu`.

If it doesn't work on your Deck, please do the following:

1. Start `sdgyrodsu` service
2. Start an emulator that uses `sdgyrodsu`
3. Close the emulator
4. Wait 15 seconds.
5. Open konsole and run following command:

    $HOME/sdgyrodsu/logcurrentrun.sh > sdgyrodsu.log

6. This will generate `sdgyrodsu.log` file. Attach it here.

You can also inspect the file, there should be messages like below:

- Server: New client subscribed. IP: ... Port: ...
- Server: No packet from client for some time. IP: ... Port: ...

Maybe there is another client that connects to `sdgyrodsu` that you don't know about?

subworx, 2022-10-04:
Pays off to think first. I only checked `top` while some switch emulator was running but not during steam games, thus gyro was running all the time, at least whenever I looked.

Sorry to bother you 🙏

closed 2022-10-04 by subworx