kmicki/SteamDeckGyroDSU #27: Steam Controller Support

Rambus200, 2022-10-10:
Will there be steam controller support or compatibility added in? If not, could you provide instructions how to make work https://github.com/FrogTheFrog/steam-gyro-for-cemuhook work in conjunction with the SteamDeckGyroDSU?

There is some demand for seeing the steam controller gyro work, as well as the deck gyro, or at least, have both the mentioned apps function at the same time.

Thanks!

kmicki, 2022-10-10:
Generally it should work at the same time if you configure a different custom port in steam-gyro-for-cemuhook. But there as some caveats apparently (see #25). As I mentioned there, the easiest would be to fork this sdgyrodsu and modify it to provide support for Steam Controller at a different port. I can't say when I'll be able to do that though.

Supreeeme, 2022-10-11:
[sc-controller](https://github.com/Ryochan7/sc-controller) has cemuhook support.

closed 2022-11-10 by kmicki