kmicki/SteamDeckGyroDSU #42: For enhancememt: Batocera Support

ruineux, 2023-03-23:
Hello,

If this can be done under batocera environment that would be great

brunocalmon, 2023-04-23:
The only missing thing to make it perfect.

git-developer, 2023-08-28:
I'm maintaining the repository [batocera-extra](https://github.com/git-developer/batocera-extra) with addons for motion and touch control in batocera. I just found your project and added support for [sdgyrodsu in a feature branch](https://github.com/git-developer/batocera-extra/tree/feature/sdgyrodsu/sdgyrodsu#readme).  Unfortunately I don't own a Steam Deck, so it is untested. Maybe someone here is interested? The documentation can be found [here](https://github.com/git-developer/batocera-extra/tree/feature/sdgyrodsu#batocera-extra).