kmicki/SteamDeckGyroDSU #47: Doesn't work with VirtualHere USB Server

Sa6o90, 2023-04-29:
I'm trying to use my SteamDeck just as a controller for Yuzu running on PC.

I connect my SD to my PC via the VirtualHere USB.
When the VirtualHere USB Server is stopped or the client is not connected, SteamDeckGyroDSU works and the Yuzu can receive motion data. The moment VirtualHere is started and connected, the motion sensor stops and Yuzu can't connect to cemuhook anymore (doesn't receive motion data). It looks like something is blocking it, but I can't figure it out.

kmicki, 2023-04-29:
VirtualHere hijacks the entire USB device that provides buttons, axes and gyro input, so it's not available anymore for sdgyrodsu to read gyro data.

If virtualhere client runs on Linux it might be possible to use sdgyrodsu on a client machine. If it's Windows then right now I can't offer any workaround.

Sa6o90, 2023-04-29:
Yes, the PC machine uses Linux as well. I will be happy to know more about the workaround. :)
Thanks!

Sa6o90, 2023-04-29:
Never mind, I make it to work the way you said. I just downloaded SteamDeckGyroDSU to my PC. Connected my SD with PC via VirtualHere, then I typed "sudo /path/to/sdgyrodsu" in the terminal of my PC and now everything works.
Thanks again. :)

closed 2023-04-29 by Sa6o90