kmicki/SteamDeckGyroDSU #56: Not Functional over Lan On CEMU 2.0

mav1o1, 2023-06-26:
CEMU shows the controller as an option when following the instructions listed in https://github.com/kmicki/SteamDeckGyroDSU/issues/36#issuecomment-1367514023
However, motion input is not registered by CEMU itself.  A test of the workaround shown in https://github.com/kmicki/SteamDeckGyroDSU/issues/45#issuecomment-1511202067 has no results, although I am testing it over steam remote play.

mav1o1, 2023-06-26:
For more context on my use case, I am attempting to stream cemu to the steam deck using GLOSI, and at the same time using this to relay gyro input.

kmicki, 2023-07-28:
You have to provide Cemu with an IP address of the deck for motion. I think it's somewhere in motion settings. Did you do that?

Also, the workaround from #45 has to be used (if it still works at all). 

You can also test with [PadTest](https://files.sshnuke.net/PadTest_1011.zip) on your host PC to check if it's issue on Cemu's or Deck's side.

kmicki, 2023-10-15:
No response - closing the issue.

closed 2023-10-15 by kmicki