kmicki/SteamDeckGyroDSU #59: DSUGyro won't install, neither via EmuDeck nor via bash command

l0wk37, 2023-10-16:
Sadly, as of yesterday, I am not able to reinstall GyroDSU after swapping my SSD and setting EmuDeck back up again.

I set my sudo pw and tried installing via Emudeck - nothing happens after pressing the install button.

Pasting the bash command into my konsole (bash <(curl -sL https://raw.githubusercontent.com/kmicki/SteamDeckGyroDSU/master/pkg/update.sh)) gives me the following error:

(deck@steamdeck ~)$ bash <(curl -sL https://raw.githubusercontent.com/kmicki/SteamDeckGyroDSU/master/pkg/update.sh)
Grabbing latest release...
Latest release downloaded.
Extracting files...
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of SteamDeckGyroDSUSetup.zip or
        SteamDeckGyroDSUSetup.zip.zip, and cannot find SteamDeckGyroDSUSetup.zip.ZIP, period.

Can I somehow install the zip manually?

kmicki, 2023-10-16:
Hello. I forgot to attach the binary package with the latest release.
My bad. Thank you for bringing this issue to my attention.

It should work now. Please try again.

kmicki, 2023-10-17:
There is another problem with the latest release which makes it not functional on current stable version of SteamOS (#60) - just for your information. If you had this problem also, right now install and update scripts point to the v1.14 release again that should work properly.

closed 2023-10-18 by kmicki