kmicki/SteamDeckGyroDSU #60: GLIBCXX_3.4.31 not found

mindset-tk, 2023-10-17:
Hello, I ran update.sh and the sdgyrodsu service will not start. When I run the log run command I get:

```
(deck@steamdeck ~)$ ~/sdgyrodsu/logcurrentrun.sh 
Oct 16 20:56:05 steamdeck sdgyrodsu[26821]: /home/deck/sdgyrodsu/sdgyrodsu: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.31' not found (required by /home/deck/sdgyrodsu/sdgyrodsu)
```

I am running the latest (October 16 update) beta SteamOS.  When I run `strings /usr/lib/libstdc++.so.6 | grep GLIBCXX` the latest version I see is GLIBCXX_3.4.30. I am unsure how to safely update this file, usually it's updated by a steamOS update.

mindset-tk, 2023-10-17:
I updated to the steam preview channel and it does work now. Perhaps there should be a note to the effect on the readme?

Thank you.

LouisTech, 2023-10-17:
I am having the same issue. Thank you for the workaround.

kmicki, 2023-10-17:
Thank you for reporting the issue. The reason is the v1.15 release was built using the preview version of SteamOS. I marked it as pre-release so update scripts point to v1.14 again until I manage to build the v1.15 against the stable version of the OS.

kmicki, 2023-10-18:
The binary for version v1.15 has been rebuilt in an Arch Linux environment with latest packages from mirrors configured in SteamOS 3.4 (i.e. stable version). It depends on GLIBCXX_3.4.30 now instead GLIBCXX_3.4.31.

The release v1.15 has been restored and will be installed using update scripts.

kmicki, 2023-10-18:
It works on the most recent SteamOS from Stable channel.

closed 2023-10-18 by kmicki

mindset-tk, 2023-10-18:
Thanks for the quick turnaround!