kmicki/SteamDeckGyroDSU #68: Sudachi Emulator Support

JaydedCompanion, 2024-06-07:
Hi! Is Sudachi currently supported? Afaik, Sudachi is a fork of Y*** so I assumed it would be supported out-of-the-box. I've tried manually copying over the gyro config from Yuzu and when I press the "Test" button on the config panel it says "Successfully received data from the server", but the gyro controls aren't working (so far I've only tested it in Tears of the Kingdom, but GyroDSU works flawlessly on Y*** with the same save file, etc.) Let me know if I need to provide any further info!

ts95, 2024-07-20:
I have the same problem with BoTW

ts95, 2024-07-24:
I got it to work after following this tutorial https://www.youtube.com/watch?v=q-ehLPk9HXI

In game mode while running a Switch game using Sudachi, you need to connect a keyboard and press F11, then open the controller settings. Unless you're using an external monitor with a higher resolution, the "Motion 1" button at the bottom won't be visible at all, so you have to use the tab key to select it (and you won't be able to see it), then you press enter, then you shake the Steam Deck.

[image]

JaydedCompanion, 2024-08-02:
This worked! Thanks for the advice and tutorial link.

To add to @ts95 's solution: you can access the bottom of the config panel without using a keyboard or external monitor by going to the game shortcut's properties (click the little gear icon to the right of the "Play" button, then select "Properties..." from the menu) and under "Shortcut" change "Game Resolution" to something higher than 1080p (I did 2560×1600). This will render the game on a virtually larger screen and thus will shrink the GUI, allowing you to see the "Motion 1" button. If you can't see Sudachi's menu bar after doing this, just click where the menu bar options _would_ be and the dropdown will appear, it just might not render for some reason. 

I also recommend switching the shortcut resolution back to default once you're done, since I assume upscaling will consume even more resources even if Sudachi is running at the native Switch resolution.

closed 2024-08-02 by JaydedCompanion